sparkfun/L6470-AutoDriver #15: Making the library compatible with Teensy

PheiPheiPhei, 2015-04-03:
Hi!

[I came across a simple fix that will make the library compatible with Teensy microcontrollers.](https://forum.pjrc.com/threads/28094-L6470-AutoDriver-stepper-driver-and-the-Teensy-3-1?p=69752&viewfull=1#post69752) Simply declaring all chars explicitly as signed will fix it.

ToniCorinne, 2016-03-09:
Hi -we'd love it if you added a pull request with this fix to the main library repository [here](https://github.com/sparkfun/SparkFun_AutoDriver_Arduino_Library). That will allow us to give you credit for the fix. 

Thanks!

PheiPheiPhei, 2016-03-09:
Hi!

I'm sadly a pretty big noob with Git and don't know how to do that. But go
ahead, I don't insist on credit. :)

christophepersoz, 2016-04-04:
Hello guys,

I'm trying to make the library working with a simple example (run, stop, back run), but it seems that something is going wrong just after sending the goToDir command. As soon as the stepper start to run until the position and after it reach the position the loop() completely stop.
Do you have this kind of issue too ?

Thanks

christophepersoz, 2016-04-25:
In order to make the library completely compatible with Teensy I had to change the SparkFunAutoDriverCommands.cpp in the same way that @PheiPheiPhei did on xferParam() and signed all the char in the file. That way, when you call GoTo() command or any other referring to SPIXfer() the behavior is now OK.

I hope it could help.

mhord, 2016-08-16:
Fixed during refactoring by changing data type to int8_t (aka signed char) from char. Thanks for the tip!

closed 2016-08-16 by mhord