sparkfun/L6470-AutoDriver #20: dSPIN_support:165 error

castfxman, 2016-02-19:
Hello,

I have used your example with success and recent reinstalled my 1.5.3 IDE and am getting a new error on line 165 of dSPIN_support it looks like

`  SPI.setClockDivider(SPI_CLOCK_DIV16); // or 2, 8, 16, 32, 64` 

Any ideas where this error could be coming from?  Thanks!

mgrusin, 2016-02-22:
It looks like setClockDivider() may have been depreciated: https://www.arduino.cc/en/Reference/SPISetClockDivider. We apologize for the inconvenience; the newest Arduino IDEs have been breaking a lot of things.

Thanks for the error report, we'll look into it!

castfxman, 2016-02-22:
Yes, in 1.5.8 version did not get this error, I tried to not include the SPI_CLOCK_DIV16 line in the 1.5.3 version which I think pumped unregulated hertz into the boards motor drivers.  I did in fact fry/smoke/burnout a couple L6470 drivers by eliminating that SPI _Clock_div16 line, but it is properly declared in 1.5.8.  SPI.h file.  So I'm using that or later from now on.

closed 2016-02-22 by castfxman