sparkfun/L6470-AutoDriver #26: AutoDriver ABS_POS problems

philk557, 2016-11-30:
About two months ago I ordered an L6470 AutoDrive V13 board from SparkFun to use with a machine tool rotary table. I have gotten the board to work with an Arduino Mega and a 34 vdc power supply.

However the position count from ADboard.getParam (ABS_POS) has problems:

1.     Even with the board set to full stepping, ADboard.configStepMode(STEP_FS), 
       ABS_POS increments by about 128 microsteps instead of full steps.

2.     The count appears to be modulo 2**16 unsigned instead of 2**21 two’s complement.

For example (the decimal numbers are 2**16 unsigned):

When ABS_POS is incremented from  0x007FFF (32,767), ABS_POS is set to 0x3F8000 (32,768).
When ABS_POS is incremented from  0x3FFFFF (65,535), ABS_POS is set to 0x000000.
When ABS_POS is decremented from 0x000000,             ABS_POS is set to 0x3FFFFF (65,535)
When ABS_POS is decremented from 0x3F8000 (32,768), ABS_POS is set to 0x007FFF (32,767).

ADboard.getPos() interprets the count as 2**21 two’s complement.

When  the count reaches 0x007FFF (32,767), it rolls over to 0xFFFF8000 (-32,768)..

Following are portions of a monitor log file of readings once a second.

get_pos is the value return by long getPos(); first HEX then DEC.
ABS_POS is the value returned by unsigned long getParam(ABS_POS), first HEX the DEC
&0x00FFFF is the ABS_POS value masked by &0x00FFFF, first HEX then DEC.

  10:2:22 Run FWD 1.00 steps/sec for 20 secs
  get_pos: 7EB9 32441  ABS_POS: 7EB9  &0x00FFFF 7EB9 32441
  get_pos: 7F3B 32571  ABS_POS: 7F3B  &0x00FFFF 7F3B 32571
  get_pos: 7FBE 32702  ABS_POS: 7FBE  &0x00FFFF 7FBE 32702
  get_pos: FFFF8040 -32704  ABS_POS: 3F8040  &0x00FFFF 8040 32832
  get_pos: FFFF80C2 -32574  ABS_POS: 3F80C2  &0x00FFFF 80C2 32962
  get_pos: FFFF8144 -32444  ABS_POS: 3F8144  &0x00FFFF 8144 33092

  get_pos: FFFFFE87 -377  ABS_POS: 3FFE87  &0x00FFFF FE87 65159
  get_pos: FFFFFF09 -247  ABS_POS: 3FFF09  &0x00FFFF FF09 65289
  get_pos: FFFFFF8B -117  ABS_POS: 3FFF8B  &0x00FFFF FF8B 65419
  get_pos: D 13  ABS_POS: D  &0x00FFFF D 13
  get_pos: 8F 143  ABS_POS: 8F  &0x00FFFF 8F 143
  get_pos: 112 274  ABS_POS: 112  &0x00FFFF 112 274

10:17:54 Run REV 1.00 steps/sec for 20 secs
  get_pos: 174 372  ABS_POS: 174  &0x00FFFF 174 372
  get_pos: F1 241  ABS_POS: F1  &0x00FFFF F1 241
  get_pos: 6F 111  ABS_POS: 6F  &0x00FFFF 6F 111
  get_pos: FFFFFFED -19  ABS_POS: 3FFFED  &0x00FFFF FFED 65517
  get_pos: FFFFFF6B -149  ABS_POS: 3FFF6B  &0x00FFFF FF6B 65387
  get_pos: FFFFFEE9 -279  ABS_POS: 3FFEE9  &0x00FFFF FEE9 65257

  get_pos: FFFF8144 -32444  ABS_POS: 3F8144  &0x00FFFF 8144 33092
  get_pos: FFFF80C1 -32575  ABS_POS: 3F80C1  &0x00FFFF 80C1 32961
  get_pos: FFFF803F -32705  ABS_POS: 3F803F  &0x00FFFF 803F 32831
  get_pos: 7FBD 32701  ABS_POS: 7FBD  &0x00FFFF 7FBD 32701
  get_pos: 7F3B 32571  ABS_POS: 7F3A  &0x00FFFF 7F3A 32570
  get_pos: 7EB8 32440  ABS_POS: 7EB8  &0x00FFFF 7EB8 32440

mhord, 2016-11-30:
Give it a try now. I fixed a logic error in the shifting of incoming SPI data that only affected parameters of greater than 16 bits.

philk557, 2016-11-30:
> <https://github.com/sparkfun/L6470-AutoDriver/issues/26#issuecomment-264023109>view
> it on GitHub, or

Hello Mike

Great!  You fixed it.

Do I download a new version of the AutoDriver library to get the fix?

Thanks

Phil

mhord, 2016-12-01:
You can either download it, clone it, or pull the changes. Downloading is probably easiest if you aren't already doing a lot of stuff via github.

philk557, 2016-12-02:
Hi Mike

I deleted the old AutoDriver library, download 
the AutoDriver ZIP file from GitHub and added the 
new library.  I didnt see any change notice in 
the header so I assumed I got your fix.

There are still problems.

In the following logs, the first get_pos number 
is HEX, the second is DEC and the third is the 
difference from the previous reading DEC.  All 
get_pos numbers are printed as LONG.  The ABS_POS is HEX.

1.  ADboard.getPos()is still reporting about 130 
steps per full step, ADboard.configStepMode(STEP_FS).

11:9:40 Run FWD 1.00 steps/sec for 10 secs
11:9:40   get_pos: 1 1 1  ABS_POS: 3
11:9:42   get_pos: 87 135 134  ABS_POS: 87
11:9:43   get_pos: 109 265 130  ABS_POS: 109
11:9:44   get_pos: 18C 396 131  ABS_POS: 18C
11:9:45   get_pos: 20E 526 130  ABS_POS: 20E
11:9:46   get_pos: 290 656 130  ABS_POS: 290
11:9:47   get_pos: 312 786 130  ABS_POS: 313
11:9:48   get_pos: 395 917 131  ABS_POS: 395
11:9:49   get_pos: 417 1047 130  ABS_POS: 417
11:9:50   get_pos: 499 1177 130  ABS_POS: 49A
11:9:51   get_pos: 51C 1308 131  ABS_POS: 51C
elapsed ms= 11141

2.      Since AB_POS is a 22 bit register with a 
signed number, I expected 2FFFFF to overflow to 0?  Not to keep on incrementing

11:12:41 Run FWD 400.00 steps/sec for 30 secs
11:12:51   get_pos: 2D2109 2957577 52226  ABS_POS: 2D2143
11:12:52   get_pos: 2DECDC 3009756 52179  ABS_POS: 2DED16
11:12:53   get_pos: 2EB8F4 3062004 52248  ABS_POS: 2EB92D
11:12:54   get_pos: 2F850B 3114251 52247  ABS_POS: 2F8545
11:12:55   get_pos: 305120 3166496 52245  ABS_POS: 305159
11:12:56   get_pos: 311D28 3218728 52232  ABS_POS: 311D61
11:12:57   get_pos: 31E928 3270952 52224  ABS_POS: 31E962
11:12:58   get_pos: 32B526 3323174 52222  ABS_POS: 32B560

11:13:20   get_pos: 3DF997 4061591 11  ABS_POS: 3DFA50
11:13:21   get_pos: 3EBD37 4111671 50080  ABS_POS: 3EBD70
11:13:22   get_pos: 3F893A 4163898 52227  ABS_POS: 3F8973
11:13:23   get_pos: 5547 21831 -4142067  ABS_POS: 557C
11:13:24   get_pos: 12153 74067 52236  ABS_POS: 12185
11:13:25   get_pos: 1ED64 126308 52241  ABS_POS: 1ED99
11:13:26   get_pos: 2B973 178547 52239  ABS_POS: 2B9A7

3.      Following is what happened when 
decrementing past 0.  I expected the next count 
to be minus.  It looks like the minus bit wasnt propagated to the left.

11:40:29   get_pos: 18B 395 -130  ABS_POS: 18B
11:40:30   get_pos: 108 264 -131  ABS_POS: 108
11:40:31   get_pos: 86 134 -130  ABS_POS: 86
11:40:32   get_pos: 4 4 -130  ABS_POS: 4
11:40:33   get_pos: 3FFF81 4194177 4194173  ABS_POS: 3FFF81
11:40:34   get_pos: 3FFEFF 4194047 -130  ABS_POS: 3FFEFF
11:40:35   get_pos: 3FFE7D 4193917 -130  ABS_POS: 3FFE7C

mhord, 2016-12-05:
1. Can you post your code somewhere? I can't duplicate the stepping issue. What does the actual motor seem to be doing? If you move forward at 1step/s, does it move a full revolution in the expected time?
2. Overflow of signed integers is left to the compiler author's imagination; there's no standard for it, so anything that happens is fair game. It looks like if you go far enough past -2^21 eventually something weird happens and you end up with a large positive integer. This appears to be how the chip does it, based on my observations of the data coming directly from the chip.
3. D'oh! I had commented out the sign propagation code while working on the earlier fix. It's back in place now. Could you download the library and give it a try?

philk557, 2016-12-05:
Hi Mike

When I downloaded the Zip file and tried to" Add 
ZIP library", Arduino IDE says "Specified 
folder/zip file does not contain a valid library".

I answer your questions later.

Thanks

Phil

philk557, 2016-12-05:
Hi Mike

I'm using a 200 steps/rev stepper motor.  If I run 1 step/sec for 200 
secs, it make a full rev.  Other speeds are consistent.

I'll try to make a project in GitHub and put the code there.

At 09:48 AM 12/5/2016, you wrote:
>Can you post your code somewhere? I can't duplicate the stepping 
>issue. What does the actual motor seem to be doing? If you move 
>forward at 1step/s, does it move a full revolution in the expected time?

philk557, 2016-12-05:
Hi Mike

I just created a respository on GitHub:

<https://github.com/philk557/Rotary-Table-Controller>philk557/Rotary-Table-<https://github.com/philk557/Rotary-Table-Controller>Controller

At 09:48 AM 12/5/2016, you wrote:
>Can you post your code somewhere?

mhord, 2016-12-05:
Hm. Using the same settings as you are, I'm definitely *not* seeing a step count problem. If I set the motor to move for 200s at 1step/s, I get ~200 steps on the position at the end of it. As I understand your problem, you see ~25600 steps as the position, right?

philk557, 2016-12-05:
Right, in the following log I get 26,117.

Hope this helps.

Thanks for sticking with this.

Phil

Here's the log:

14:2:21 Sketch started

RUN TABLE
14:2:41 Run FWD at 0.25 degrees/sec for 1 secs
14:2:41 Run FWD 10.00 steps/sec for 1 secs
14:2:41 microSteps: 0 Pos: 0.00 deg  0 steps   Plus 0.00 deg 0.00 steps
14:2:41 get_pos: 8 8 8  ABS_POS: 1A
14:2:42 get_pos: 543 1347 1339  ABS_POS: 544
14:2:43 microSteps: A5A Pos: 0.50 deg  20 steps   Plus 0.25 deg 10.02 steps
elapsed ms= 2118

14:27:24 Sketch started

RUN TABLE
14:27:48 Run FWD at 0.03 degrees/sec for 200 secs
14:27:48 Run FWD 1.00 steps/sec for 200 secs
14:27:48 microSteps: 0 Pos: 0.00 deg  0.00 steps   Plus 0.00 deg 0.00 steps
14:27:48 get_pos: 1 1 1  ABS_POS: 3
14:27:49 get_pos: 87 135 134  ABS_POS: 87
14:27:50 get_pos: 109 265 130  ABS_POS: 109
14:27:51 get_pos: 18B 395 130  ABS_POS: 18B
14:27:52 get_pos: 20D 525 130  ABS_POS: 20D
14:27:53 get_pos: 28F 655 130  ABS_POS: 28F
14:27:54 get_pos: 311 785 130  ABS_POS: 311
14:27:55 get_pos: 393 915 130  ABS_POS: 393
14:27:56 get_pos: 415 1045 130  ABS_POS: 415
14:27:57 get_pos: 497 1175 130  ABS_POS: 497
14:27:58 get_pos: 519 1305 130  ABS_POS: 519
14:27:59 get_pos: 59B 1435 130  ABS_POS: 59B
14:28:0 get_pos: 61D 1565 130  ABS_POS: 61E
14:28:1 get_pos: 6A0 1696 131  ABS_POS: 6A0
14:28:2 get_pos: 722 1826 130  ABS_POS: 722
14:28:3 get_pos: 7A4 1956 130  ABS_POS: 7A4
14:28:4 get_pos: 826 2086 130  ABS_POS: 826
14:28:5 get_pos: 8A8 2216 130  ABS_POS: 8A8
14:28:6 get_pos: 92B 2347 131  ABS_POS: 92B
14:28:7 get_pos: 9AD 2477 130  ABS_POS: 9AD
14:28:8 get_pos: A2F 2607 130  ABS_POS: A2F
14:28:9 get_pos: AB1 2737 130  ABS_POS: AB1
14:28:10 get_pos: B33 2867 130  ABS_POS: B33
14:28:11 get_pos: BB5 2997 130  ABS_POS: BB5
14:28:12 get_pos: C37 3127 130  ABS_POS: C38
14:28:13 get_pos: CBA 3258 131  ABS_POS: CBA
14:28:14 get_pos: D3C 3388 130  ABS_POS: D3C
14:28:15 get_pos: DBE 3518 130  ABS_POS: DBE
14:28:16 get_pos: E40 3648 130  ABS_POS: E40
14:28:17 get_pos: EC2 3778 130  ABS_POS: EC2
14:28:18 get_pos: F44 3908 130  ABS_POS: F44
14:28:19 get_pos: FC6 4038 130  ABS_POS: FC7
14:28:20 get_pos: 1049 4169 131  ABS_POS: 1049
14:28:21 get_pos: 10CB 4299 130  ABS_POS: 10CB
14:28:22 get_pos: 114D 4429 130  ABS_POS: 114D
14:28:23 get_pos: 11CF 4559 130  ABS_POS: 11CF
14:28:24 get_pos: 1251 4689 130  ABS_POS: 1251
14:28:25 get_pos: 12D3 4819 130  ABS_POS: 12D3
14:28:26 get_pos: 1355 4949 130  ABS_POS: 1356
14:28:27 get_pos: 13D8 5080 131  ABS_POS: 13D8
14:28:28 get_pos: 145A 5210 130  ABS_POS: 145A
14:28:29 get_pos: 14DC 5340 130  ABS_POS: 14DC
14:28:30 get_pos: 155E 5470 130  ABS_POS: 155E
[... 199 lines cut ...]

/* ------------------------------------------------------------------
  *  Read stepper motor position from the AutoDriver Board.
*/

long  AD_abs_pos()
{
   long abs_pos =0;
   long get_pos =0;
   long twos    =0;

   time_stamp();
   get_pos =ADboard.getPos();
   Serial.print("get_pos: ");
   Serial.print(get_pos, HEX);
   Serial.print(" ");
   Serial.print(get_pos);
   Serial.print(" ");
   Serial.print(get_pos-prev_pos);
   prev_pos=get_pos;
   abs_pos =ADboard.getParam(ABS_POS);
   Serial.print("  ABS_POS: ");
   Serial.println(abs_pos, HEX);
   return (abs_pos);
}

mhord, 2016-12-06:
Found it. On line 957 you issue the "resetDev()" command, which is going to reset all the registers to their power-on state. The chip starts up with 128 microsteps/step as the default. Try commenting out that line and see what it gives you.

philk557, 2016-12-07:
Hello Mike

Thank you for finding that!  I appreciate it.

After moving the resetDev ahead of all the config 
settings, the steeper wouldn't run.  The 
setOCThreshold(OC_1500mA) was too low and OC 
shutdown occurred.  I changed the threshold to 2250 ma and now it works fine.

Let me suggest that in 
https://learn.sparkfun.com/tutorials/getting-started-with-the-autodriver---v13#arduino-library---operation 
you add to

void resetDev();

Equivalent to toggling the reset pin, but 
performed through an internal software register 
control on the L6470 chip. If you dont want to 
use a pin for reset, this option is for you.

a note that parameters are reset to the values 
listed on page 40 of the L6470 datasheet.

I still can't add the latest ZIP library file 
from the AutoDriver GitHub respository.  Arduino 
IDE Add ZIP library says "Specified folder/zip 
file does not contain a valid library".

Thanks again

Phil

mhord, 2016-12-12:
I'll make a note in the tutorial about what exactly that reset function does.

Do you know how to manually install the library? Just extract the archive and drag the folder into the "libraries" directory inside your sketchbook folder. See if it works doing that and I'll see if I can't figure out why the IDE doesn't like it.

mhord, 2016-12-12:
Also, make sure you're downloading the zip file of *just* the library, not the library and hardware files both. [Here is the link](https://github.com/sparkfun/SparkFun_AutoDriver_Arduino_Library) to the library repository.

MarcusWolschon, 2017-02-09:
Is this issue still open or did someone forget to close it?

mhord, 2017-02-09:
It should have been closed. Sorry about that!

closed 2017-02-09 by mhord

philk557, 2017-02-09:
Sorry, forgot I needed to close it. Mike H. helped me resolve the problem.

Phil Keys