sparkfun/L6470-AutoDriver #28: Mechanical drawing

MarcusWolschon, 2017-02-09:
A simple mechanical drawing that provides the location of the mounting holes and the board dimensions would be very nice.
Not everyone has EAGLE or knows how to use it to get a DXF these from the .brd file.

mhord, 2017-02-09:
We've (globally) elected not to do this since there is a free version of Eagle available and we include measurements in the board design files. We realize that it doesn't work for everyone, but neither will dxf, svg, or any other format we choose. Eagle is simply the least common denominator.

closed 2017-02-09 by mhord

MarcusWolschon, 2017-02-13:
So for non-personal or commercial use the user is expected to buy Eagle or take their own meassurements? This is rediculus.

mhord, 2017-02-13:
[image]

Here's a PNG of the mechanical layout. You bring up an interesting point regarding licensing of EAGLE. I'll bring the topic up for debate again.