sparkfun/L6470-AutoDriver #29: Issues with "GetPos()"

Vachefolle, 2017-02-15:
Hi,

I've an issue with the function GetPos(). Effectively, if we see this picture :
[image]

When I move the motor, the getPos function display the current position of motors. If I move the motor 100 steps, getPos display 100 steps. If I move 200 steps, getPos display 72 steps, etc. Between 0 and 127, getPos display 0 to 127. But between 128 and 200, we can see 0 to 72...
 
I really don't know where the problem comes from.

The last version of AutoDriver library is : 1.3.2.
Motor : [http://www.omc-stepperonline.com/gear-ratio-511-planetary-gearbox-with-nema-14-stepper-motor-14hs130804spg51-p-138.html](url)
AutoDriver : [https://www.sparkfun.com/products/11611](url)
OS : Ubuntu 16.04

Somebody had the same problem and fixed it ?

Thanks !

PS : my code : 

```
void setupPIN()
{
	//Setup wiringPi
	wiringPiSetup();

	//Setup PIN RESET/BUSY/ENDPOS A & B
	pinMode(PIN_RESET, 				OUTPUT); //PIN RESET
	pinMode(PIN_BUSY, 				INPUT);  //PIN BUSY
	pinMode(PIN_LASER, 				OUTPUT);  //PIN LASER
	pinMode(PIN_A_ENDPOSITION_A, 	INPUT);
	pinMode(PIN_A_ENDPOSITION_B, 	INPUT);
	pinMode(PIN_B_ENDPOSITION_A, 	INPUT);
	pinMode(PIN_B_ENDPOSITION_B, 	INPUT);

	//Open port CE0
	digitalWrite(PIN_CS, HIGH);

	//Reset Board A & B
	digitalWrite(PIN_RESET, LOW);
	digitalWrite(PIN_RESET, HIGH);
}

void setupBOARD_b()
{
	boardA.configSyncPin(PIN_BUSY, 0);
	boardA.configStepMode(STEP_FS);
	boardA.setOscMode(INT_16MHZ_OSCOUT_16MHZ);
	boardA.setMinSpeed(0);
	boardA.setMaxSpeed(500);
	boardA.setAcc(3000);

	boardB.configSyncPin(PIN_BUSY, 0);
	boardB.configStepMode(STEP_FS);
	boardA.setOscMode(INT_16MHZ_OSCOUT_16MHZ);
	boardB.setMinSpeed(0);
	boardB.setMaxSpeed(500);
	boardB.setAcc(3000);
}
int main()
{
	setupPIN();
	setupBOARD_b();
	long step;
	while (1)
	{
		boardB.resetPos();
		cout << "Input a step : "; cin >> step;
		boardB.move(1,step);
		while(boardB.busyCheck()){
		}
		cout << "test = " <<  boardB.getPos() << endl;
		delayy(1000);
		boardB.move(0,step);
	}
	return 0;
}
```

Vachefolle, 2017-02-15:
But when I tell it to move 200 steps, motor moves 200 steps (because it crossed 2 times the distance of 100 steps). So, the problem doesn't come from motors... :/

mhord, 2017-02-20:
I can't duplicate this under Arduino 1.8.1. A simple application which turns the motor on for ten seconds and prints the position ten times a second yields the results I'd expect.

I'm going to suggest that you get the library directly from the library repository, just to make sure it is, in fact, the most recent version. [Here's a link to that page.](https://github.com/sparkfun/SparkFun_AutoDriver_Arduino_Library)

Vachefolle, 2017-02-20:
Thanks for your answer. I use a raspberry Pi 3 ^^

And I already got back the last version. But don't worry. I found a solution to have the correct position. Increment and decrement a global variable which get back the movement : 

**Exemple :**
Move left 100 steps (direction : 1, getPosA +=  100) then move right 50 steps ( direction : 0, getPosA -= 50) then etc. 

Same for the second motor with getPosB.

mhord, 2017-02-20:
Okay, I'm glad you have a workaround. It looks to me like some kind of overflow error or something, since it breaks at 128, maybe there is a place where the data flows through an int8_t datatype?

Vachefolle, 2017-02-21:
I thought about the same problem (kind of overflow). Maybe but where ?
Do you exactly want my project (part of autodriver library)? Maybe you'll find something that pass before my eyes ! :/

It isn't in the rush, but I'll reuse this board within a short time and I'll find this problem again... 

In any case, thank for your answers !

closed 2017-03-08 by Vachefolle