sparkfun/L6470-AutoDriver #30: Uploading Example Sketch to Arduino Mega 2560

MeijinGandalf, 2017-03-02:
I was having issues uploading the SparkFunGetSetParamTest sketch to the Arduino Mega 2560. I found that any Serial.print or Serial.println that had three exclamation points in a row "!!!" cause the upload to hang. I was able to comment those lines out and get the sketch to upload.  Changing those to two in a row also fixed the uploading issue.  Have you seen this issue yourself?

mhord, 2017-03-02:
I haven't tried it on the Arduino Mega. Perhaps something in the bootloader uses the exclamation point as an escape character?

MeijinGandalf, 2017-03-17:
I am not sure.  When looking through a bunch of forums i found another person that had the same problem when trying to upload a different problem. When uploading to the Uno R3 there was no problem.

mhord, 2017-03-17:
My tech support team tells me that this is, in fact, a known issue with the bootloader on the Mega. It treats three exclamation points as an escape sequence. The only fix is to remove one or more exclamation points.

closed 2017-03-17 by mhord