sparkfun/L6470-AutoDriver #34: goto/gotodir functions do not work

wandergeek, 2019-04-01:
Hello,

I seem to be having issues getting the `goTo` and `goToDir` functions to work. When the function runs, the motor moves quickly and then stops responding. I see serial messages from my program indicating that the micro is still alive.

I can also report that the `run` function works fine. Source examples for both below:

## Working
```arduino
void loop() {
      if (dir == 1) {
        Serial.println("going fwd");
        boardA.run(FWD, 100);
      } else {        
        Serial.println("going bwd");
        boardA.run(REV, 100); 
    }
      Serial.println("zzzz 2 secs");
      delay(2000);
      dir*=-1;
      Serial.println("soft stopping");
      boardA.softStop();
      Serial.println("checking busy");
      while (boardA.busyCheck()) {
          Serial.print("busy pos");
          Serial.println(boardA.getPos());
      }
}
```
## Not working

```arduino
void loop() {
      if (dir == 1) {
        Serial.println("going fwd");
        boardA.goToDir(FWD,400);
      } else {        
        Serial.println("going bwd");
        boardA.goToDir(REV,400); 
    }
      Serial.println("zzzz 2 secs");
      delay(2000);
      dir*=-1;
      Serial.println("soft stoppping");
      boardA.softStop();
      Serial.println("checking busy");
      while (boardA.busyCheck()) {
          Serial.print("busy pos");
          Serial.println(boardA.getPos());
      }
}
```

wandergeek, 2019-04-07:
Is there any extra info I can provide? :)

edspark, 2019-04-22:
Sorry for the very delayed response. Shortly after I looked after your issue, I was called away suddenly from work because my wife went into labor. I'm looking into it now.

edspark, 2019-04-23:
Hello @wandergeek, as far as I can tell your code runs fine for me. I made a slight modification switching the `boardA.busyCheck()` function with your delay so that I could get a readout of the stepper motor's position as it was turning, and _then_ delaying for 2 seconds.  Please see the attached serial output: 

[image]

I think seeing all of your code may help me determine what the issue is, and stating the power you're giving your motors. Below I've linked the code in its' entirety that I'm using to run the motors, perhaps that would be more helpful. 

```C++
#include <SparkFunAutoDriver.h>
#include <SPI.h>

int CSPin = 10; 
int boardPos = 0; 
int resetPin = A4;
int busyPin = A5; 
int dir = 1; 

AutoDriver boardA(boardPos, CSPin, resetPin, busyPin);

void setup() {

  Serial.begin(115200);
  Serial.println("Motors and stuff.");  
  boardA.SPIPortConnect(&SPI);
  SPI.begin();
  SPI.setDataMode(SPI_MODE3);

}

void loop() {

  if (dir == 1) {
    Serial.println("Forward...");
    boardA.goToDir(FWD, 400);
  } 
  else {        
    Serial.println("Backward...");
    boardA.goToDir(REV, 400); 
  }

  while (boardA.busyCheck()) {
      Serial.print("Board Position: ");
      Serial.println(boardA.getPos());
  }

  Serial.println("Stopping...");
  delay(2000);
  dir *= -1;
  boardA.softStop();

}
```

edspark, 2019-04-23:
I'm going to close this for the time being, please reopen if you have more questions.

closed 2019-04-23 by edspark